RoboDK/RoboDK-API #20: robot not responsive to keyboard

avpai, 2019-02-27:
Hi,
I recently got the license for RoboDK and I wanted to control a KUKA arm using keyboard. I am using the provided sample. 
```
# This macro allows moving a robot using the keyboard
# Note: This works on console mode only, you must run the PY file separately
#
# More information about the RoboDK API here:
# https://robodk.com/doc/en/RoboDK-API.html
# Type help("robolink") or help("robodk") for more information

from robolink import *    # API to communicate with RoboDK
from robodk import *      # basic matrix operations
RDK = Robolink()

# Arrow keys program example

# get a robot
robot = RDK.Item('', ITEM_TYPE_ROBOT)
if not robot.Valid():
    print("No robot in the station. Load a robot first, then run this program.")
    pause(5)
    raise Exception("No robot in the station!")

print('Using robot: %s' % robot.Name())
print('Use the arrows (left, right, up, down), Q and A keys to move the robot')
print('Note: This works on console mode only, you must run the PY file separately')

# define the move increment
move_speed = 10

import msvcrt
while True:  
    key = ord(msvcrt.getch())
    move_direction = [0,0,0]
    print(key)
    if key == 75:
        print('arrow left (Y-)')
        move_direction = [0,-1,0]
    elif key == 77:
        print('arrow right (Y+)')
        move_direction = [0,1,0]
    elif key == 72:
        print('arrow up (X-)')
        move_direction = [-1,0,0]
    elif key == 80:
        print('arrow down (X+)')
        move_direction = [1,0,0]
    elif key == chr(113):
        print('Q (Z+)')
        move_direction = [0,0,10]
    elif key == 97:
        print('A (Z-)')
        move_direction = [0,0,-1]

    # make sure that a movement direction is specified
    if norm(move_direction) <= 0:
        continue

    # calculate the movement in mm according to the movement speed
    xyz_move = mult3(move_direction, move_speed)

    # get the robot joints
    robot_joints = robot.Joints()

    # get the robot position from the joints (calculate forward kinematics)
    robot_position = robot.SolveFK(robot_joints)

    # get the robot configuration (robot joint state)
    robot_config = robot.JointsConfig(robot_joints)

    # calculate the new robot position
    new_robot_position = transl(xyz_move)*robot_position

    # calculate the new robot joints
    new_robot_joints = robot.SolveIK(new_robot_position)
    if len(new_robot_joints.tolist()) < 6:
        print("No robot solution!! The new position is too far, out of reach or close to a singularity")
        continue

    # calculate the robot configuration for the new joints
    new_robot_config = robot.JointsConfig(new_robot_joints)

    if robot_config[0] != new_robot_config[0] or robot_config[1] != new_robot_config[1] or robot_config[2] != new_robot_config[2]:
        print("Warning!! Robot configuration changed!! This will lead to unextected movements!")
        print(robot_config)
        print(new_robot_config)

    # move the robot joints to the new position
    robot.MoveJ(new_robot_joints)
    #robot.MoveL(new_robot_joints)
```

But the robot is not responsive to the keyboard press. If I give a specific value to 'key' and then run the program, the robot moves. Any help or suggestions will be greatly appreciated.

RoboDK, 2019-02-27:
You should run this program in console mode (not within RoboDK as a macro). This means that you must save the file as a PY file and double click it to run it (or run it from Python IDLE). Then, the keyboard will work.

avpai, 2019-02-27:
```
C:\Users\avpai\Desktop>py kuka1.py
Traceback (most recent call last):
  File "kuka1.py", line 8, in <module>
    import robolink     # API to communicate with RoboDK
  File "C:\Python34\Lib\robolink\__init__.py", line 3, in <module>
    import pkgutil
  File "C:\Python34\Lib\pkgutil.py", line 4, in <module>
    import importlib
  File "C:\Python34\Lib\importlib\__init__.py", line 34, in <module>
    _w_long = _bootstrap._w_long
AttributeError: module 'importlib._bootstrap' has no attribute '_w_long'
```
I tried that and I keep getting the above error. I copied the library files into my python directory but its still there.

RoboDK, 2019-02-27:
You may need to install the RoboDK API for Python manually:
C:/Python34/Scripts/pip install robot

Then, run the kuka1.py file:
C:/Python34/python.exe kuka1.py

avpai, 2019-02-27:
Thank you I got it running, but the robot changes its position for a split second and then goes back to its original position. I am using KUKA-LBR-iiwa-14-R820. Is this an issue with the code? I tried with other robots too but had the same experience.

RoboDK, 2019-02-27:
We currently do not support a robot driver for the IIWA robots. We can help you set it up, more information here:
https://robodk.com/doc/en/Robot-Drivers.html#DriverCustom

avpai, 2019-02-27:
Could you please tell me what robot arm are currently supported. I just wanted to run some simulations in the software and I am not working with any actual arm at the moment.

avpai, 2019-02-28:
Guys,

I tried this example from the python API documentation. I still got the same problem. Robot moves for a split second then goes back to it original pose.

```
from robolink import *                  # import the robolink library        
RDK = Robolink()                        # connect to the RoboDK API (RoboDK starts if it has not started
robot  = RDK.Item('', ITEM_TYPE_ROBOT)  # Retrieve the robot

x_move = 2
y_move = 3
z_move = -1

# get the current robot joints
robot_joints = robot.Joints()

# get the robot position from the joints (calculate forward kinematics)
robot_position = robot.SolveFK(robot_joints)

# get the robot configuration (robot joint state)
robot_config = robot.JointsConfig(robot_joints)

# calculate the new robot position
new_robot_position = transl([x_move,y_move,z_move])*robot_position

# calculate the new robot joints
new_robot_joints = robot.SolveIK(new_robot_position)
if len(new_robot_joints.tolist()) < 6:
    print("No robot solution!! The new position is too far, out of reach or close to a singularity")
    continue

# calculate the robot configuration for the new joints
new_robot_config = robot.JointsConfig(new_robot_joints)

if robot_config[0] != new_robot_config[0] or robot_config[1] != new_robot_config[1] or robot_config[2] != new_robot_config[2]:
    print("Warning! Robot configuration changed: this may lead to unextected movements!")
    print(robot_config)
    print(new_robot_config)

# move the robot to the new position
robot.MoveJ(new_robot_joints)
#robot.MoveL(new_robot_joints)

```
I tried running this on all the robot arm provided in the installation folder. Am I missing something here?
I am working on an important project and I need to get the simulation working. End goal is to control the robot using IK (by giving x,y,z of the end effector). I tried with the keyboard and this. I am using the 
KUKA-LBR-iiwa-14-R820 for just simulation purposes. If it something wrong with this arm kindly suggest which arm I can use to do the simulation. Any help and suggestions will be appreciated. Thanks :)

avpai, 2019-02-28:
Ok, guys. I tried it on another computer and I was working. But the 2nd computer doesn't have a license and its on trail period.  I want to run this on my own computer not the 2nd one, so what should I do? Will my license key be still valid if I re-install the software? please advise.

avpai, 2019-03-02:
[image]

RoboDK, 2019-03-04:
I recommend you to reinstall RoboDK with the Python option activated (unchecked by default if you reinstall RoboDK). You should not have problems with the update if you have a license and your maintenance is not expired. In any case, using RoboDK just for that shouldn't require a license.

kennetao, 2019-04-11:
We have the same exact problem. We are using RoboDK for our bachelor project. It works on some computers but on others it does not. We have tried reinstalling RoboDK with the python 3.4 option but this does not fix the issue.
Any advise would be greatly appreciated.

avpai, 2019-04-11:
@kennetao 
I am sorry but I was not able to solve the problem.  I have opened a thread in the RoboDK Forum.

https://robodk.com/forum/Thread-roboDK-not-working-properly

If you could comment there, it will really help both of us to find a solution.

kennetao, 2019-04-12:
Hi.

We had the same issue with RoboDK. It worked on 1 out of 3 computers we tested with. We solved the issue by copying the RoboDK folder from the working pc to the pc's that it did not work on.
It looks like it is a version issue. The working version is 3.5.5 64bit. The non working version is 3.6.2 64bit.

avpai, 2019-04-12:
Hi,
Thats great news. Could you please tell me where you downloaded the older version? Is it possible to share the setup?

kennetao, 2019-04-12:
The older version was installed almost a half a year ago on that pc. So we guess that was the version available from RoboDK at the time it was installed. We guess searching for Version 3.5.5 in your broswer might be helpfull.

kennetao, 2019-04-12:
This thread might be helpfull:
https://robodk.com/forum/Thread-Download-older-versions-of-RoboDK

RoboDK, 2019-04-15:
@avpai @kennetao Thank you for noticing. It was a RoboDK bug (with the SolveIK function). A workaround would be to provide a guess for the robot joints. For example: robot.SolveIK(pose, joints).

In any case, we solved this bug with the latest version: 
https://robodk.com/downloads/Install-RoboDK-64-v3.6.3.exe

closed 2019-04-15 by RoboDK