RoboDK/RoboDK-API #23: [Python] Use Qt instead of tkinter if available

davidmurray, 2019-05-15:
Hello,

Right now, using this library forces tkinter to be included (which increases the size of program's executable!) for the file management functions.

Would there be a way to use Qt instead if PySide2 is already being imported into the project?

Regards,
David

RoboDK, 2019-05-16:
You can remove the dependency of TKinter in robodk.py, iIt should be very easy. Everything related to TKinter is towards the end of the module. You can just make a copy of the robodk.py file and remove that section.

davidmurray, 2019-05-17:
Yes, I am aware of this, but I was talking about including Qt support directly in the class.

David

RoboDK, 2019-05-17:
You can add Qt support separately (it does not need to be through RoboDK API) and you can remove the TKinter dependency in robodk.py. It should not be complicated and you can still have full access to the RoboDK API without TKinter or Qt.

closed 2019-08-14 by RoboDK