RoboDK/RoboDK-API #36: [BUG] AddShape specifiying the addTo parameter does not work

sancelot, 2019-12-22:
Using AddShape in either python or c++ language does not permit to specify the addTo parameter.

when setted in, the function replies with following message:
**WARNING: Invalid shape, a 3xN (or 6xN) list of points must be provided (N must be multiple of 3)
Invalid item provided: The item identifier provided is not valid or it does not exist.**

The following code reproduces the problem :

```
from robodk import *
from robolink import *

# Octahedron vertices 
A = [ 0.17770898,  0.72315927,  0.66742804]
B = [-0.65327074, -0.4196453 ,  0.63018661]
C = [ 0.65382635,  0.42081934, -0.62882604]
D = [-0.17907021, -0.72084723, -0.66956189]
E = [-0.73452809,  0.5495376 , -0.39809158]
F = [ 0.73451554, -0.55094017,  0.39617148]

# Octadron faces by groups of vertices
SHAPE = [E, B, A,
        E, D, B,
        E, C, D,
        E, A, C,
        F, A, B,
        F, B, D,
        F, D, C,
        F, C, A,
]

##### Add the shape as an object in RoboDK

RDK = Robolink() # Initialize the RoboDK API
part = RDK.Item("Part")
if part.Valid():
    new_object = RDK.AddShape(SHAPE,part)
```

nubi86, 2019-12-22:
Is "Part" an object? If so, I'm unable to reproduce this warning.

Your Octahedron is very small. I recommend you to scale it as shown in the following code:

`    new_object = RDK.AddShape(SHAPE, part) # this works
    new_object = RDK.AddShape(Mat(SHAPE).tr() * 200, part) # this scales x100`

sancelot, 2019-12-23:
Part is a "Frame", I added using this code (c)
 
```
Item repere_piece = RDK->getItem("Part");
if (!repere_piece.Valid())
	{
	repere_piece = RDK->AddFrame("Part", NULL);
	}

// same reply with this code 
    Item shape = RDK->AddShape(ptr, &repere_piece,  false, &c);

```

sancelot, 2019-12-23:
I managed to insert the shape next to the frame using . so what is supposed being used "addTo"  for?

```
 Item shape = RDK->AddShape(ptr, nullptr,  false, &c);
 shape.setParent(repere_piece);
```

nubi86, 2019-12-23:
Excellent, this is the best way to go. 

The AddTo parameter allows you to add a shape to an existing object. So you would be adding the geometry to the same object. You can not add a shape to a reference frame.

You can also add geometry to a robot link. 

For example, if you want to add geometry to the 3rd link of a robot you can do:
```
Item robot = RDK->getItem("Your Robot");
Item robot_link = robot.ObjectLink(3);
RDK->AddShape(ptr, &link, false, color);
```

closed 2020-09-04 by RoboDK