RoboDK/RoboDK-API #39: [Python] Update() replies with incorrect simulation time

sancelot, 2020-01-07:
Using the next snippet, that generates a program with different targets and speed, the update func is used to know the estimated simulation time.

Unfortunately, it replies correctly only for the first program call.
To obtain the right simulation time, I have to uncomment this code (and this not wanted in my case....):
```
   prog.RunProgram()
   while prog.Busy():
         time.sleep(0.1)
```

python program I used:
```
# # -*- coding: utf-8 -*-
from robolink import *    # API to communicate with RoboDK for simulation and offline/online programming
from robodk import *      # Robotics toolbox for industrial robots

#
# https://robodk.com/doc/en/PythonAPI/robodk.html#robodk.Mat
#
import time
import_install("numpy", "numpy")
import numpy as np

def AddTarget(name,pos):
    item = RDK.Item(name)
    itemparent = RDK.Item("World")
    if not item.Valid():
        item = RDK.AddTarget(name, itemparent, itemrobot=0)
    #print(item)
    m = item.Pose()
    #print(m)
    #print(type(m))
    m=Mat().eye()
    #m=transl(pos[0], pos[1], pos[2])
    # xyz wpr deg
    m = m.Offset(pos[0],pos[1],pos[2],pos[3],pos[4],pos[5])
    item.setPose(m)
    

# Any interaction with RoboDK must be done through RDK:
RDK = Robolink()

# Select a robot (popup is displayed if more than one robot is available)
robot = RDK.ItemUserPick('Select a robot', ITEM_TYPE_ROBOT)
if not robot.Valid():
    raise Exception('No robot selected or available')

tool = RDK.Item('flange')
tool.AttachClosest()
tool.DetachAll()
HomeJoints=[-66.18,-17.813,-3.961,0,-86.039,66.18]
robot.setJoints(HomeJoints)

# Turn off rendering (faster)
RDK.Render(True)

HomeTarget=[445,-1008,1000,180,0,0]
ErwinHome = AddTarget("ErwinHome",HomeTarget)
home=RDK.Item("ErwinHome")
# definit la vitesse lineaire
robot.setSpeed(40,-1,-1,-1), # linear mm/sec
init = True
result =""
no=0
mode = 'L' # 'L' Lineaire / Sinon Joint MoveJ
#RDK.setSimulationSpeed(100)

for speed in [40,400,1000,2000,4000]:
 for rounding in [-1,0,10,25,50,100]:
  for dx in [10,100,500,1000,1500]:
    oldprog = RDK.Item("AutoProgram")
    if oldprog.Valid():
        oldprog.Delete()
    #robot.setPose(RDK.Item("T1").Pose())
    nomprog = 'Program_dx'+str(dx)+'_cnt_'+str(rounding)+'_speed_'+str(speed)
    prog = RDK.AddProgram("AutoProgram")
    if init:
        if mode == 'L':
            prog.MoveL(home)
        else:
            prog.MoveJ(home)
        init = False
    # Hide program instructions (optional, but faster)
    prog.ShowInstructions(True)
    prog.setSpeed(speed,speed)
    prog.setRounding(rounding)
    Home = Pose(445,-1008,1000,180,0,0)

    T1=Pose(0,-dx,0,0,0,0)
    M=Home*T1
    AddTarget("T1",Pose_2_Fanuc(M))
    t1 = RDK.Item("T1")
    if mode == 'L':
        prog.MoveL(t1)
    else:
        prog.MoveJ(t1)
    T2=Pose(dx,-dx,0,0,0,0)
    M=Home*T2
    AddTarget("T2",Pose_2_Fanuc(M))
    t2 = RDK.Item("T2")
    if mode == 'L':
        prog.MoveL(t2)
    else:
        prog.MoveJ(t2)
    T=Pose(dx,-dx,dx,0,0,0)
    M=Home*T

    AddTarget("T3",Pose_2_Fanuc(M))
    t3 = RDK.Item("T3")
    if mode == 'L':
        prog.MoveL(t3)
    else:
        prog.MoveJ(t3)

    T=Pose(dx,0,dx,0,0,0)
    M=Home*T
    AddTarget("T4",Pose_2_Fanuc(M))
    t4 = RDK.Item("T4")
    if mode == 'L':
        prog.MoveL(t4)
    else:
        prog.MoveJ(t4)

    T=Pose(0,0,dx,0,0,0)
    M=Home*T
    AddTarget("T5",Pose_2_Fanuc(M))
    t5 = RDK.Item("T5")
    if mode == 'L':
        prog.MoveL(t5)
    else:
        prog.MoveJ(t5)
    if mode == 'L':
        prog.MoveL(home)
    else:
        prog.MoveJ(home)
    check_collisions =  False
#    prog.RunProgram()
#    while prog.Busy():
#         time.sleep(0.1)
    update_result = prog.Update(check_collisions)

#    print("Valid instructions ",update_result[0])
#    print("valid_ratio ",update_result[3])
#    print("message ",update_result[4])
    s = "%6.3f" % update_result[1]
    s=s.replace('.',',') # pauvre conversion pour excel en mode french
    print(s)
    r = "dx %d speed: %d mm/s CNT:%d temps cycle %f sec.\n" % (dx,speed,rounding,update_result[1])
    #print(r)
    result +=r
#    time.sleep(30)

print("END")
print(result)
```
wrong results without RunProgram():
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000
 1,500
15,000
75,000
150,000
225,000

right results with RunProgram():

1,660
15,160
75,160
150,160
225,160
1,660
15,160
75,160
150,160
225,160
1,500
15,000
75,000
150,000
225,000
1,500
15,000
75,000
150,000
225,000
1,500
15,000
75,000
150,000
225,000
1,500
15,000
75,000
150,000
225,000
0,980
3,098
9,100
16,600
24,100
0,980
3,098
9,100
16,600
24,100
0,150
1,500
7,500
15,000
22,500
0,150
1,500
7,500
15,000
22,500
0,150
1,500
7,500
15,000
22,500
0,150
1,500
7,500
15,000
22,500
0,980
3,098
6,928
10,000
13,000
0,980
3,098
6,928
10,000
13,000
0,060
0,600
3,000
6,000
9,000
0,060
0,600
3,000
6,000
9,000
0,060
0,600
3,000
6,000
9,000
0,060
0,600
3,000
6,000
9,000
0,980
3,098
6,928
9,798
12,000
0,980
3,098
6,928
9,798
12,000
0,030
0,300
1,500
3,000
4,500
0,030
0,300
1,500
3,000
4,500
0,030
0,300
1,500
3,000
4,500
0,030
0,300
1,500
3,000
4,500
0,980
3,098
6,928
9,798
12,000
0,980
3,098
6,928
9,798
12,000
0,015
0,150
0,750
1,500
2,250
0,015
0,150
0,750
1,500
2,250
0,015
0,150
0,750
1,500
2,250
0,015
0,150
0,750
1,500
2,250

RoboDK, 2020-01-07:
It looks like there is around 10% discrepancy... This does not look normal.
Do you have the RDK file to reproduce this issue?
Can you try calling Update twice?

sancelot, 2020-01-08:
Using Update()  twice does not help. rdk file is enclosed
 
[demo_0801.zip](https://github.com/RoboDK/RoboDK-API/files/4034105/demo_0801.zip)

closed 2020-09-04 by RoboDK