RoboDK/RoboDK-API #46: PoseFrame() raises Exception

FlixMa, 2020-06-17:
When I try to use the PoseFrame() method on a target, I receive the following error:

`Exception: Invalid item provided: The item identifier provided is not valid or it does not exist.`

This is the code I used:
```
    target = link.Item('my_target', robolink.ITEM_TYPE_TARGET)
    assert target.Valid()
    pose_relative_to_robot_base = target.PoseFrame()
```

I tried this using rdk v4.2.3.15578 and v5.0.0-17259 in conjunction with pypi package `robodk` version 4.2.3 and 5.0.0.
But it doesn't look like an error in the pypi package, as the command `G_Frame` is not understood.

Any Ideas?

As a workaround I moved the robot into the origin of the world coordinate system so I can use `PoseAbs()`.

RoboDK, 2020-06-17:
You should use target.Pose()
PoseFrame is the robot base frame (if the item is a robot).

FlixMa, 2020-06-17:
Okay thanks.

I would like to move the robot to a given target with a specific configuration, e.g. (front, elbow up, flip). Therefore I'm searching for all configurations to get to a given item' pose using [`SolveIK_All()`](https://robodk.com/doc/en/PythonAPI/robolink.html?highlight=solveik#robolink.Item.SolveIK_All):
> `pose (Mat)` – pose of the robot flange with respect to the robot base frame

I guess I need to perform two steps:
- I need to subtract the tool offset from my desired target.
- The flange pose needs to be relative to the robots base frame.
- run `SolveIK_ALL(desired_flange_pose)`

Is there an easy way of doing this?
Maybe the undocumented optional parameters are what I am searching for? (`SolveIK_All(pose, tool=None, reference=None)`)

RoboDK, 2020-09-04:
Yes, you can provide the tool and reference to SolveIK_All if the pose is a target.

In any case, it is easy to calculate the pose of the flange with respect tot he robot by applying this calculation:
```
target = (target pose)
pose = reference * target * invH(tool)
```

closed 2020-09-04 by RoboDK