RoboDK/RoboDK-API #51: Error in C++ API

Tokrat, 2020-08-27:
Hi, 
I started to use your API a few days ago for a C++ program and I have several issue:

1. The function __AddStation__ doesn't work
  - The argument __name__ is not even used in __robodk_api.cpp__
  - The function doesn't create anything, it just print in console: __RoboDK API ERROR:  ""__

> Once I get this error, all function of the API will give the same error until I disconnect / reconnect

2. I often get the error __Communication problems with the RoboDK API__ and after __RoboDK API ERROR:  ""__ without any more information or error. Even if first I got __The RoboDK API is connected__, I don't know why and restart the software reset the problem, I cannot debug it without more informations.

3. __SolveIK__ just crash (core dump) without any information, I started to modify __robodk_api.cpp__ to debug but it could take a while.

4. In all your example, you add the robot, the tool etc.. and only use the pointer afterward. But if you read from an existing .RDK file, you will not get any pointer but the real item. I don't really understand why it use this pattern, if the pointer is not required to manipulate an object, why not use a reference, which is simpler and safer. And if the pointer is required, the API will simply not work when opening an existing scene and editing the object since we do not get a pointer but an Item (ex: AddFile)

5. Why is there "Finish" and "Disconnect", according to your API, it's exactly the same. Is one depreciated ?

---

The last part is not really a problem but more a personal note about the API, it does not seems to follow the usual standard, it looks more like a port from C than a code written for C++, since it doesn't seems to be done I give you my feedback:
- I understand that there is some Qt overhead for the Matrix etc. But it bother me that I have to convert each "std::string" to the QString format. It's not necessary since you don't even use the QString, you parse it __toUtf8__ before sending it, so I don't really see the point.
- The use of raw pointer is more common in C, the new standard is to use the Smart pointer, which avoid any memory handling / pointer movement.
- In your API, you sometime ask for pointer, sometime for the object. It would be better to just call the reference for most functions, it did not feel really consistent when using the API. 
- There is no apparent Linting standard, it's sometimes Pascal case, sometimes Camel case, sometimes even Snake case. This caused a lot of correct / recompile during the use of the API, which was annoying.

> Some of my remark could be wrong or even a choice you made on purpose. I just wanted to notified what could be improve in my point of view

---

Lastly, is there a way to try mulptiple "MoveL_Test" in separate thread simultaneously ?

RoboDK, 2020-09-04:
Thank you for your feedback. Issue 1 should be fixed now. 

Issues 2 and 3 could be related to issue 1. If you have a way to reproduce them it would help if you can provide an example.

I'm not sure I understand issue 4. RoboDK pointers change when you restart RoboDK. These pointers unique until you delete your objects. Therefore, these pointers must be retrieved by name when you start RoboDK unless you build your cell by loading files.

I understand the API naming can be confusing for some programmers. We tried to keep the naming as uniform as possible among different programming languages. I agree that you'll find a mixture of Python, C and C++ naming convention among others.

It is not possible to run multiple MoveL_Test simultaneusly using the same RoboDK instance. On the other hand, you can run mutliple instances of RoboDK.

Tokrat, 2020-09-04:
1. Issue is indeed fixed ! 
2. Issue still appear from time to time but it's just the message **RoboDK API ERROR: ""** The API works anyway, and it's not all the time, I wonder if it's not something like a timeout. It's not a big problem, I will come back to you if I succeed to identify more precisely when it comes (I didn't a lot of testing, it's not impossible that it's my fault, I will check more)
3. Issue was due to asynchronous request of the API, so error from my side. You could maybe add an error if the API receive request from multiple thread for the next person doing the same mistake ? When I stopped multi threading the error disappeared.

4. This issue was not very "clear" sorry, I've meant to say that in all your example you set everything (robot, tool, station, etc...) using pointer, and a lot of function asked for a pointer, but when I create the station myself in the API, I had to have an object **Item** and not ***Item** since the fonction "addrobot" for exemple return an **Item**, so it was not clear if the pointer is required or if the ref to the object would be enough, since my program was not working yet I was not sure. After some test I know now that the pointer are not required at all, I don't have any in my program and just use **&robot** for example when the function ask for the pointer, everything works and it should be exactly the same in term of performance.
Let's say it's just that when a program ask specifically for a pointer, using just a reference to the object will not work because of the pattern, here it works (which is better), so it's just that I would probably get rid of the pointer in the example and use the reference inside the parameter called in the API function, it's just a minor detail but it was unclear.

---

I understand you wanted to keep the same naming convention everywhere, it's just that it would be a lot better to have one naming convention for one API, I don't mind if the naming is not standard, just that it's consistant.

Ok I understand for the multiple instances, but could they all work with the same licence ? (I think so but just to be sure)

RoboDK, 2021-04-19:
I'm sorry we missed your question. I confirm that multiple instances can be used with one license. 
I recommend you to contact us via our forum:
https://robodk.com/forum/
Albert