RoboDK/RoboDK-API #56: howto set robot configuration from api ?

sancelot, 2020-10-27:
Hi,
I would like being able to setup my robot configuration for each target.
I have not found a function for this.
Steph

sancelot, 2020-10-28:
This is  not really efficient (in terms of computing time)
I have almost 50 targets to setup ......

I will try it. But is there a better way when we know configuration to apply.?

sancelot, 2020-10-28:
I tried the following snippet , It replies with **[[ ]]** :worried: 

`# # -*- coding: utf-8 -*-
# This macro shows an example to draw a polygon of radius R and n_sides vertices using the RoboDK API for Python
from robolink import *    # API to communicate with RoboDK for simulation and offline/online programming
from robodk import *      # Robotics toolbox for industrial robots
import math

# Any interaction with RoboDK must be done through RDK:
RDK = Robolink("localhost",port=20501)

# Select a robot (popup is displayed if more than one robot is available)
robot = RDK.ItemUserPick('Select a robot', ITEM_TYPE_ROBOT)
if not robot.Valid():
    raise Exception('No robot selected or available')

Target =RDK.Item("Target_10")
if Target.Valid():
    pose = Target.Pose()
    joints = robot.SolveIK_All(pose)

print(joints)
`

sancelot, 2020-10-29:
@c3dff , this not really what I need finally, 

I know for each points of my FANUC robot I want to setup : L/R U/D T/B configs.

Here I may receive joints positions .....only . I can't filter anything

RoboDK, 2020-10-30:
You can do Mat.list() to return a one dimensional list of doubles. Example:

`    pose = Target.Pose()
    joints = robot.SolveIK_All(pose)
    joints = joints.list()
    valid_solution = len(joints) == 0
    print(joints)
    `

RoboDK, 2020-10-30:
To properly find your desired configuration you should use JointsConfig, you can use the first 3 values as [Rear, Lower, Flip] flags. The 4th value is used for turn count on certain robots. This shows an example:

```
# Retrieve all solutions for a given pose:
all_solutions = robot.SolveIK_All(pose, toolpose, framepose)
joints = []

# Iterate through each solution
for j in all_solutions:
    # Retrieve flags as a list for each solution
    conf_RLF = robot.JointsConfig(j).list()

    # Breakdown of flags:
    rear  = conf_RLF[0] # 1 if Rear , 0 if Front
    lower = conf_RLF[1] # 1 if Lower, 0 if Upper (elbow)
    flip  = conf_RLF[2] # 1 if Flip , 0 if Non flip (Flip is usually when Joint 5 is negative)

    # Look for a solution with Front and Elbow up configuration
    #if conf_RLF[0:2] == [0,0]:
    if rear == 0 and lower == 0:
        print("Solution found!")
        joints = j
        break
```

sancelot, 2020-10-30:
Thanks, but the joint values I can get with SolveIk_All() are not the same as what is displayed in Robodk 

[image]

[image]

c3dff, 2020-10-30:
To remove the possibility of pose reference error, I wrote this generic filter (not the prettiest code, but should work). It takes a target ITEM as an input and applies the filter in the same way the GUI allows (Both=None as args to fd, ud and fn). for those args, the value mirrors the output of `JointsConfig` so  [0,0,0] means [front, upper arm and non-flip]. 
I'd really wish to have a low-level command to the solver. Having to solve for all is very inefficient for large target counts.

```
def filter_configurations(target, fb=None, ud=None, fn=None):
    robot = target.getLink()
    target_pose_abs = target.PoseAbs()

    flange_pose = target_pose_abs * robodk.invH(robot.PoseTool())
    pose = robodk.invH(robot.PoseAbs()) * flange_pose

    joints_list = robot.SolveIK_All(pose)
    njoint = robot.Joints().size(dim=0)

    valid_configs = []
    for i in range(joints_list.size(dim=1)):
        sol = joints_list[0:njoint, i]
        # conf_RLF (int list)
        # robot configuration as a list of 3 ints: [REAR, LOWER-ARM, FLIP]
        # [0,0,0] means [front, upper arm and non-flip] configuration.
        conf_rlf = robot.JointsConfig(sol)

        results = list(map(int, conf_rlf.list()))

        valid = True

        # Checks front-rear
        if fb is not None:
            test = results[0] == int(fb)
            valid = valid and test

        # Checks up-down
        if ud is not None:
            test = results[1] == int(ud)
            valid = valid and test

        # Checks flip-nonflip
        if fn is not None:
            test = results[2] == int(fn)
            valid = valid and test

        if valid:
            valid_configs.append(sol.list())

    return valid_configs
```

sancelot, 2020-11-02:
Ok, we are not so far from what I need.

Now, to activate my configuration, I have to set point as a Joint Target instead of cartesian and give joint positions corresponding to desired configuration?

closed 2021-08-22 by RoboDK