RoboDK/RoboDK-API #60: C# - RoboDK top level window was not found...

klemikaze, 2020-12-01:
Hello.
I cloned the repo, tried to run the project from Visual Studio (tried versions 2017 and 2019) and got System.FormatException at RoboDK.cs line 584, as the previous line
`var mainWindowId = Command("MainWindow_ID");`
returns "RoboDK top level window was not found..." instead of int descriptor of the window. This happens both if the RDK is opened previously before the project is run, or if the RDK is not running.

I couldn't find much help on the toppic, the thread
https://robodk.com/forum/Thread-Problem-in-installing-running-the-RoboDk-software-Windows-8-64bit
didnt help.

klemikaze, 2020-12-01:
Previous instance of the RoboDK IDE was left open somewhere (OS warned me that existing opened instance is preventing the PC to reboot, couldn't find it otherwise), causing socket communication from 2nd instance to get aborted and disconnected from the host - socket is opened, but the "VerifyConnection" method finds out that the connection is closed - but desn't handle this state.

RoboDK, 2020-12-01:
Can you try using this command:
RDK.Command("QtMainWinID")
If it fails, what operating system are you using?
I recommend you to wait for RoboDK startup and make sure you don't run RoboDK hidden or without the UI.

klemikaze, 2020-12-02:
I wasn't able to reproduce the issue since reboot. Sorry for false alarm.