RoboDK/RoboDK-API #62: Bug in RoboDK::WindowID()

francoislauzon, 2021-03-12:
I think there is a bug in the C++ API method RoboDK::WindowID().
The current method from the repositery is like this:
```
quint64 RoboDK::WindowID(){
    qint64 window_id;
    if (window_id == 0) {
        QString response = Command("MainWindow_ID");
        window_id = response.toInt();
    }
    return window_id;
}
```
But **window_id** is not initialized.
I changed the code to:

```
quint64 RoboDK::WindowID(){
    QString response = Command("MainWindow_ID");
    quint64 window_id = response.toInt();
    
    return window_id;
}
```
And now it is working.

RoboDK, 2021-03-12:
Thank you for your feedback!
We just fixed this issue.
Albert

francoislauzon, 2021-03-12:
That was fast, thank you!

closed 2021-03-12 by francoislauzon