RoboDK/RoboDK-API #63: command line option '-HIDDEN' is not working in the API

francoislauzon, 2021-03-12:
I am using the '-HIDDEN' command line option with the C++ API and it is not working, RoboDK is not starting in background and the application show up.  I use the following constructor:

`    RoboDK_API::RoboDK *rdk_=new RoboDK_API::RoboDK("",kRoboDKPragmaPort,"/NOSPLASH /HIDDEN");`

The '-NOSPLAHS' option is working.
When I run the command directly with **robodk.exe -HIDDEN** it is not working either.

Am I doing something wrong or is it suppose to work?

I use the latest C++ API and RoboDK v5.2.1

Thanks,
Francois.

vin-eet-singh, 2021-03-12:
Hi francoislauzon,
Thank you for your feedback.
Instead of:
`RoboDK_API::RoboDK *rdk_=new RoboDK_API::RoboDK("",kRoboDKPragmaPort,"/NOSPLASH /HIDDEN");`
You can use:
`RoboDK_API::RoboDK *rdk_=new RoboDK_API::RoboDK("",kRoboDKPragmaPort,"/NOSPLASH -NOUI");`

Let me know how it goes.

francoislauzon, 2021-03-12:
If I use the '-NOUI' argument, when I later show RoboDK using the API the mainwindow is empty.

What I am trying to acheive is to start RoboDK using the API and then integrate it as a widget... If you modify the Cpp-sample with the following mainwindow constructor you will be able to reproduce what I want (except there is the visual glitch at startup that I am trying to avoid):

```
MainWindow::MainWindow(QWidget *parent) :
    QMainWindow(parent),
    ui(new Ui::MainWindow)
{

    robodk_window = NULL;
    ui->setupUi(this);
    ui->widgetRoboDK->hide();
    adjustSize();

    // Start RoboDK API here (RoboDK will start if it is not running)
    ROBOT = NULL;
    RDK = new RoboDK("",-1,"/NOSPLASH /HIDDEN");

    if (!RDK->Connected()){
        qDebug() << "Failed to start RoboDK API!!";
    }
    // I have to hide RDK because '/HIDDEN' doesn't work
    RDK->HideRoboDK();

    // add a little delay, otherwise the integration will fail
    QThread::msleep(500);
    on_radIntegrateRoboDK_clicked();
}

```

The only thing I am trying to avoid using the '/HIDDEN' flag is the initial showing/hidding of RoboDK as a Top Level Window before it get integrated in the Cpp-Sample, any other ideas?

Thanks,
Francois.

vin-eet-singh, 2021-03-13:
Hi Francois,
Thank you for bringing this up. We did find a bug which will be fixed in RoboDK's next update. The next update should be in a week or so.

Do let us know if you run into any other issues.

francoislauzon, 2021-04-08:
Hello,
I have upgraded to the latest version of RoboDK yesterday.
The /HIDDEN flag is now working. 
But if I run the example as I described it previously,  RoboDK start in fullscreen and is not integrated in the Example application.
How can I start RoboDK as an integrated Qt Window at startup, is it possible?

Best regards,
François.

RoboDK, 2021-04-19:
Hi François,
I'm not sure if it possible to retrieve the ID of the main window before starting RoboDK. You may need to use a Windows API.
Albert

closed 2021-08-22 by RoboDK