RoboDK/RoboDK-API #97: small mistake on German RoboDK website

TXAE, 2022-08-03:
This is not about the API, but instead about your German website translation.

[https://robodk.com/de/](https://robodk.com/de/)

below "Roboter-Bearbeitung"

it has to be "die Roboterbahn" instead of the current "den Roboterbahn".

TXAE, 2022-08-03:
Not a mistake, but sounds weird: [https://robodk.com/de/simulation](https://robodk.com/de/simulation)

Below "1. Wählen Sie Ihren Roboter aus"
"Zugang zur RoboDK umfangreichen Roboter-Bibliothek" should be "Zugang zur umfangreichen RoboDK Roboter-Bibliothek"

TXAE, 2022-08-03:
[https://robodk.com/de/pricing](https://robodk.com/de/pricing)

Below "SCHULLIZENZ
2-JAHRES-ABONNEMENT"

"2-JAHRES-ABONNEMENT
NUR FÜR SCHULUNGSZWECKEN"

should be

"2-JAHRES-ABONNEMENT
NUR **ZU** SCHULUNGSZWECKEN"

OR

"2-JAHRES-ABONNEMENT
NUR FÜR SCHULUNGSZWECKE"

RoboDK, 2022-08-19:
Thank you for your feedback! We just fixed it

closed 2022-08-19 by RoboDK