RoboDK/RoboDK-API #99: Filters in RoboDK lib

TXAE, 2022-09-06:
Hi,

this is again a website issue and not about the the API, just like [#97](https://github.com/RoboDK/RoboDK-API/issues/97).
Since I can't find your website source on GitHub and haven't received a reply via email, I don't know where else to make you aware of it but here.

The library filters are static. This leads to you having to add a new static filter term every time you add an object to the lib with a new attribute that does not exist yet. For example, you have tools from various Brands in your lib (ATI, Aim Robotics, ...), but these brands cannot be searched via the filters on the left as the filter terms are hardcoded. Your website doesn't even allow a manual query: [https://robodk.com/library#filter?brand=ATI&type=tool](https://robodk.com/library#filter?brand=ATI&type=tool) is forwarded to [https://robodk.com/library#filter?type=tool](https://robodk.com/library#filter?type=tool).

Now you could go ahead and keep hardcoding the terms manually, but I think it would be wise to make them dynamic: Show all brands, types, axes etc. of objects you add to the lib as search terms in the filter. This could be a PR if your website code was on GH.

RoboDK, 2022-09-06:
Tools don't fall into brand filters. Tools have their own Tool type filter. Brand filters are displayed only for robot arms.

TXAE, 2022-09-06:
I'm not sure I understand. When I [filter for Tools](https://robodk.com/library#filter?type=tool), I can only filter for Kinova, Mecademic and OTC Daihen tools. But you have a lot of other tool brands like Lorch, Obara, OnRobot and so on, why can't I filter them? It's bc the filter terms are hardcoded and don't depend on lib content , no?

TXAE, 2022-09-06:
in other word: Why would tools have a brand and **not** fall into brand filters?🤔

RoboDK, 2022-09-06:
This is a good point, however, the main purpose of the library is to filter by robot brands, types, etc. You can still use the search bar filter to find your tool, or simply use Ctrl+F and search for the terms you are looking for.

RoboDK, 2022-09-06:
This is not an API issue so we are closing this thread. You can post improvement suggestions in our Forum:
https://robodk.com/forum/

closed 2022-09-06 by RoboDK