RoboDK/RoboDK-API #104: Wrong string format for double type

Aeradan, 2022-11-07:
Hello,

at line 1512 the character array bufferTemp is filled with the double values of a Mat structure x, y, z, w, p, r.
First, the lf instead of the bare f field specifier should be used, but I had problems from another point of view: my robot arm tried to reach a pose where x was in the 1e+80s, and y and z were in the 1e-300s. In this case, bufferTemp is filled with 80 characters from x alone, and when sprintf adds the values from the other members of the Mat, you get an out-of-bounds (bufferTemp is only 128 chars long).
In this case, the g field specifier should be used.
In order to respect your own style of code, I wrote:

    if ((x < 1e9) && (x > 1e-9))
        sprintf(bufferTemp, "%12.3lf,", x);
    else
        sprintf(bufferTemp, "%12.3g,", x);
    strcat(output, bufferTemp);
    if ((y < 1e9) && (y > 1e-9))
        sprintf(bufferTemp, "%12.3lf,", y);
    else
        sprintf(bufferTemp, "%12.3g,", y);
    strcat(output, bufferTemp);
    if ((z < 1e9) && (z > 1e-9))
        sprintf(bufferTemp, "%12.3lf,", z);
    else
        sprintf(bufferTemp, "%12.3g,", z);
    strcat(output, bufferTemp);
    if ((w < 1e9) && (w > 1e-9))
        sprintf(bufferTemp, "%12.3lf,", w);
    else
        sprintf(bufferTemp, "%12.3g,", w);
    strcat(output, bufferTemp);
    if ((p < 1e9) && (p > 1e-9))
        sprintf(bufferTemp, "%12.3lf,", p);
    else
        sprintf(bufferTemp, "%12.3g,", p);
    strcat(output, bufferTemp);
    if ((r < 1e9) && (r > 1e-9))
        sprintf(bufferTemp, "%12.3lf))\n", r);
    else
        sprintf(bufferTemp, "%12.3g))\n", r);
    strcat(output, bufferTemp);

I also corrected the f specifier at line 1522, 1611 and 1780.

Aeradan, 2022-11-08:
I further found that the if conditions should be further refined, and become:

`if ((fabs(x) < 1e9) && (fabs(x) > 1e-9))
`
with the introduction of the fabs() function.

ASE-Optics-Europe, 2024-02-08:
What's relevant is not the `l`, but that `%f` OR `%g` specifiers should be used depending on the values to be set into `bufferTemp`. Failing to do so will lead to odd character strings to be sprinted out, due to scientific notation being used.

vdm-dev, 2024-02-08:
> What's relevant is not the `l`, but that `%f` OR `%g` specifiers should be used depending on the values to be set into `bufferTemp`. Failing to do so will lead to odd character strings to be sprinted out, due to scientific notation being used.

Fortunately, there's no more `bufferTemp`. But there is a `svalue` that is only 20 characters long. Each `svalue` is converted from double separately and immediately written to the output.

commit 5af809e9 (2024-02-09): Merge pull request #140 from RoboDK/hotfix/RDK-2992/string-format

Fix #104
closed 2024-02-09 by vdm-dev