RoboDK/RoboDK-API #111: Problem in controlling the robotic arm from the software

SatanHunter, 2023-03-08:
Helllo everybody,

I am currently using RoboDK software, as well as the libraries for python, in order to control a Staubli TX2-90, linked to a SP2 controller.

I am able to make the python script control the simulated robotic arm (on the software). However, when I try to move the actual robotic arm, it does not work. I also tried from the software. I am able to connect to the robotic arm (I am able to retrieve the robotic arm current position), but I am unable to make it move from the software as I wish (in real time)

I found this [forum](https://robodk.com/forum/Thread-St%C3%A4ubli-Driver) detailing the same issue that I have, but with a different controller. Does anybody knows how to access the /usr/configs on the machine?

Gratefully,

Leo

sambertrdk, 2023-03-08:
Hi @SatanHunter, as this is not related to the API code itself, would you mind opening a thread on the forum instead?
I trust you have already set the run mode: https://robodk.com/doc/en/PythonAPI/robodk.html?highlight=runmode#robodk.robolink.Robolink.setRunMode
Thank you!

SatanHunter, 2023-03-08:
Hello Sam,

I hope this message finds you well,

I unfortunately d not have access to the forum (the registration process is on the way, I haven't received the confirmation code...) so I posted it on github.

Sorry for the inconvenience,

Leo

A: RoboDK/RoboDK-API
Cc: Avoni Leonardo; Mention

sambertrdk, 2023-03-09:
Moved to the forum

closed 2023-03-09 by sambertrdk