RoboDK/RoboDK-API #123: MakeProgram freeze

rwuthric, 2023-07-08:
When running [`MakeProgram`](https://robodk.com/doc/en/PythonAPI/robodk.html#robodk.robolink.Item.MakeProgram) in conditions where an error happens (e.g. the path is not writable or doesn't exist), the method call freeze.

- When running with a GUI a popup window with error message appears (very good)
- But when running in headless mode (e.g. `RoboDK/bin/RoboDK --platform minimal -NOUI`) it would be desired that the method call does not freeze and returns error in order actions can be taken from the user code side.

Same behaviour is observed  in a `try`-`except`  statement like so:
```
try:
   prog.MakeProgram(filesave)
except:
   print("[ERROR] could not save robot code file '" + filesave + "'")
```

sambertrdk, 2023-07-10:
You can use `RDK.Command("Popups", 0)` to prevent RoboDK from opening prompts. This is particulary usefull when running with the `-NOUI` option, or in Docker.

rwuthric, 2023-07-11:
That is a great option. However, does it block all pop-up windows?
It blocks indeed the pop-up of something like:
```
RDK.ShowMessage("I am a pup-up window")
```
but, at least in my case, it doesn't block the pop-up window of the error message produced by the post-processor (a window called `Python Post Processor output`).

I am running RoboDK 5.6.0.23250 under Linux Ubuntu 22.04 LTS.

rwuthric, 2023-07-13:
This doesn't seem to have any effect on error windows of the post-processors (at least not on Linux version of RoboDK). It seems to affect only the log windows.

There is as well an option `Tools->Options->Python->Show Python output` where one can select either `Always' or 'If errors` but no option like `Never` which is maybe what would needed.

sambertrdk, 2023-07-17:
We will add the `Never` for Python output. Moreover, the `-NOUI` option of RoboDK will now ensure that no pop-ups are generated. This feature will be available in our next build.

rwuthric, 2023-07-17:
Amazing. Looking forward to it.

closed 2023-07-17 by sambertrdk