RoboDK/RoboDK-API #125: Copyed Rotational Matrices in Degree and not Rad

marcopepunkt, 2023-07-19:
The highlighted section contains an error. According to the Documentation robodk.robomath.roty(ry) must recive ry in radians and not degrees. This can be confusing when copying this snippet into a Python Project.

[image]

I don't know if this is the correct place to post these errors. Please redirect me if I´m wrong here. 
Thank you.

sambertrdk, 2023-10-16:
We will add a new `Script (mm,rad)` option, thank you for your feedback!

RoboDK, 2023-10-25:
The documentation you are referring to is related to the Python API and matrix functions it provides. The User interface is using degrees in this specific example.

To use such a string using Python you could use a function like the following:

```
def String2Pose(posestring):
    from robodk import robomath
    import math
    context = {}
    context["transl"] = robomath.transl
    context["rotx"] = lambda rx: robomath.rotx(rx * math.pi / 180.0)
    context["roty"] = lambda ry: robomath.roty(ry * math.pi / 180.0)
    context["rotz"] = lambda rz: robomath.rotz(rz * math.pi / 180.0)
    # pose = eval("transl(100,0,0)*rotx(180)*roty(90)", context)
    pose = eval(posestring, context)
    return pose

pose = String2Pose('transl(0, 0, 200) * roty(-90)')

```

You can find a similar example and use case here:
https://robodk.com/forum/Thread-How-to-get-the-xyz-values-for-the-target-position-of-a-movement-from-a-program

closed 2023-10-25 by RoboDK