RoboDK/RoboDK-API #149: ERROR: Package 'robodk' requires a different Python: 3.8.10 not in '<3.5'

kbrameld, 2024-03-15:
Recently, installing  ``pip3 install robodk`` fails on Ubuntu 20.04 since 5.6.9. This worked fine with 5.6.8.

```
Collecting robodk
  Downloading robodk-5.6.9-py2.py3-none-any.whl (117 kB)
     |████████████████████████████████| 117 kB 7.1 MB/s 
ERROR: Package 'robodk' requires a different Python: 3.8.10 not in '<3.5'
```

sambertrdk, 2024-03-15:
What is the version of your Python, 3.8.10?
RoboDK 5.6.9 introduces type hints, which only works with Python 3.5+. Thus, we created two versions which are automatically selected.
Python 3.8.10 should have ">=3.5" in the setup.py, the file name should be robodk-5.6.9-py3-none-any (which is different from your trace).

See here:
https://github.com/RoboDK/RoboDK-API/blob/master/Python/setup.py#L18
https://pypi.org/project/robodk/#files

kbrameld, 2024-03-15:
Thanks for the quick response. Yes, my python version is 3.8.10. This can be reproduced in a clean ubuntu20.04 docker container via:
```
docker run ubuntu:20.04 sh -c 'apt update && apt install python3-pip -y && pip3 install robodk'
```

sambertrdk, 2024-03-21:
Thank you, I will investigate further.
For now, you can manually download and install the py3 version (https://pypi.org/project/robodk/#files) or modify the setup.py to remove the python_require.

sambertrdk, 2024-03-27:
Can you run `pip install --upgrade pip` to ensure you are using the latest version of pip?

kbrameld, 2024-04-03:
Upgrading pip to 24.0 does seem to get around the issue. It seems like the previous pip version 20.0.2 (default on Ubuntu 20.04) prefers downloading the py2.py3 version over the py3 version.

sambertrdk, 2024-04-03:
Closing with the recommendation to upgrade pip.

closed 2024-04-03 by sambertrdk

kbrameld, 2024-04-03:
I think upgrading pip is a workaround, but not a solution. Requiring users to upgrade pip leads to extra steps in user setup, CI pipelines, and the resulting pip version no longer matching the system's recommended version, potentially leading to issues elsewhere. Given Ubuntu 20.04 is a fairly popular OS (and I'm assuming is a targeted platform), I see not being able to use system-default pip to install this package a remaining issue.

sambertrdk, 2024-04-04:
Version 5.7.0 should fix this issue: https://pypi.org/project/robodk/