RoboDK/RoboDK-API #153: `robolink.Robolink.ShowMessage()` prints message in command line

sjohnson-FLL, 2024-04-11:
Pretty much the title.

The `ShowMessage` method of the `Robolink` class has a line: `print(message)` after being called. If a CLI uses the RoboDK API, this can be disruptive.

I think either this line should be removed completely or `print_msg:str=True` should be an optional argument that controls whether the message is also printed to the command line.

sambertrdk, 2024-04-17:
@RoboDK There are a couple of other functions that prints. Should we add a verbose level?

RoboDK, 2024-06-25:
We just removed the debug print for ShowMessage functions. We don't need this debug print. This function used to invoke RoboDK and we kept this print for historical reasons. 

With the next RoboDK release we'll now always print blocking popups through STDOUT.

closed 2024-06-26 by sambertrdk