RoboDK/RoboDK-API #165: C++ API, getCollisionItems bad command

dfleury2, 2025-01-30:
Hi,
in `List<Item> RoboDK::getCollisionItems(QList<int> link_id_list)` method, the underscore is missing on the command line sent to RoboDK.

    _send_Line("Collision Items");

should be 

   _send_Line("Collision_Items");

as in Python API.

Regards,

vdm-dev, 2025-01-30:
Thank you, we'll fix that soon.

vdm-dev, 2025-02-03:
The problem should be fixed. Reopen this issue again if it is not.

closed 2025-02-03 by vdm-dev

dfleury2, 2025-02-03:
Hi,
the correction seems to work but the main branch does not compile anymore due to some others errors (like using std::exception with arguments).
And there are lot of new warnings, or strange things like throwing a pointer of a new allocated exception.
I will keep my version, and follow the changes only if I them.
Regards