RoboDK/RoboDK-API #171: Mat.ToUR is broken

lxop, 2025-03-14:
As the title says. This bug appears if the rotation angle is a multiple of 180°. Example:

```cs
RoboDk.API.Mat m = RoboDk.API.Mat.FromUR(new double[]{-36.9, -146.44, 475, 2.221441469079183, -2.221441469079183, 0.0 });
double[] returned = m.ToUR();
Console.WriteLine(string.Join(", ", returned));
```

The above snippet prints `-36.9, -146.44, 475, -1, 1, 8.65927457071936E-17` . Note that the final three values don't match the original pose values.

The underlying problem is that the `{rx,ry,rz} = ` lines in the final `else` block of this method shouldn't be in there - that block should only set `mult_factor`.

The correct behaviour can be seen in the Python API implementation.

sambertrdk, 2025-03-16:
Refers to C# Mat.cs:
https://github.com/RoboDK/RoboDK-API/blob/195a9b9ef8f991edf89b89722c404bd72ef1449a/C%23/API/Mat.cs#L566
https://github.com/RoboDK/RoboDK-API/blob/195a9b9ef8f991edf89b89722c404bd72ef1449a/C%23/Example/RoboDKSampleProject/RoboDK.cs#L771

Compared to Pythhon:
https://github.com/RoboDK/RoboDK-API/blob/195a9b9ef8f991edf89b89722c404bd72ef1449a/Python/robodk/robomath.py#L1063

vdm-dev, 2025-03-21:
Hi @lxop, thank you for the message.
I have prepared a fix in a separate branch (https://github.com/RoboDK/RoboDK-API/tree/hotfix/RDK-4401/mat-ur-is-broken).
Can you check that everything is working now?

lxop, 2025-03-23:
@vdm-dev It certainly fixes my broken case, thanks, and looks like it matches the Python implementation.
It seems like this sort of thing would benefit from unit test coverage, @sambertrdk